gw-vis/kagra-vis #11: Develop better ACCs

MiyoKouseki, 2020-04-16:
From #8 

Develop better ACCs
- Takahashi-san is developping a better ACC using a shadow sensor. Some improvement of the sensor noise is expected. The driving circuit will be much simpler and robust. The feedback control can be done with KAGRA digital system. Tuning of the mechanical resonant frequency will also improve the sensitivity.

MiyoKouseki, 2020-07-27:
takahashi-san did.
https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=11774

MiyoKouseki, 2020-08-24:
@Ryutaro-T -san, Is this truly finished?

I think you and sato-san are making new LVDT coil adapter for new accelerometers.
Could you make new issues for it?

Ryutaro-T, 2020-08-25:
This task includes following items;

- [x] Research of the position sensors. (Takahashi)
- [x] Preparation of the LVDT prototype. (Sato)
- [x] Test f the ACC prototype with LVDT. (Takahashi)
- [x] Evaluation of the ACC prototype at the KAGRA site. (Miyo, Takahashi)

Ryutaro-T, 2020-08-25:
"Replacement of ACCs" should be made as new issue. Preparation of the LVDT coils is included in the issue.

Ryutaro-T, 2020-08-25:
We have already #50.

MiyoKouseki, 2020-08-25:
This issue was done.
https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=11774

closed 2020-08-25 by MiyoKouseki