gw-vis/kagra-vis #33: (My dream) Data analysis pipeline for automatic model fitting using simulated models.

MiyoKouseki, 2020-08-17:
- [ ] Create a diaggui-python interface to trigger/read active measurements.
- [ ] Create a reference library of suspension models.
- [ ] Fetch the model from the library and compare and optimize to fit the model to the measurements.
- [ ] Save fitted model to another "fitted" library.
- [ ] Use models from "fitted" library for further purpose.

terrencetec, 2020-08-24:
Per VIS meeting on 19th, Aug, I mentioned the importance of accurate models for advanced controls. While the fitted models are scattered around, I would like to propose a way to gather all models and make a way such that model fitting is fully consistent throughout and fully automatic. In such a way, we can access the models in the database with, say, 1 line of python code. This makes control design very convenient and enables the possibility of rolling optimization of control filter.

MiyoKouseki, 2020-08-25:
Yes, I totally agree with you!

Could you define how do we review this issue to close?

MiyoKouseki, 2020-08-25:
by the way, we have usefull python library "dtt2hdf" about it
> Create a diaggui-python interface to trigger/read active measurements.

but this library can read from saved dtt files. we can not execute any command from this library.

terrencetec, 2020-08-25:
> by the way, we have usefull python library "dtt2hdf" about it
> 
> > Create a diaggui-python interface to trigger/read active measurements.
> 
> but this library can read from saved dtt files. we can not execute any command from this library.

Yes, I am aware, I have been using it in my kontrol library "examples".

terrencetec, 2020-08-25:
> Yes, I totally agree with you!
> 
> Could you define how do we review this issue to close?

Not sure, this thing is still a dream to me because I am focusing on other stuff. I imagine we can test it on one suspension and then check all transfer functions to see if the fit in the database matches the measurement. And then, we can do an automatic test without supervising and check the same.

MiyoKouseki, 2020-08-25:
OK. I move this issue as Icebox in current project. 
https://github.com/orgs/gw-vis/projects/10

If we have time, we will do it in near future.