gw-vis/kagra-vis #34: Control system upgrade #2: Sensor correction optimization.

MiyoKouseki, 2020-08-17:
- [ ] Re-evaluate current sensor correction filter, redesign with Hinf synthesis.
- [ ] LMS algorithm for automatic optimization of sensor correction calibration.
- [ ] Sensor correction for other seismic noise coupled sensors, e.g. optical levers.

terrencetec, 2020-08-24:
This is probably an advanced upgrade for the sensor correction system and not mandatory for O4. I had an idea to find the correct sensor correction gain which was never possible without bias. So, that's what I would like to try and include it as part of my thesis. Also, the current sensor correction filter is somewhat random. I have been able to reduce the peak height with similar phase performance around microseism, so I would like to make another one using H2/H-infinity later.

MiyoKouseki, 2020-08-25:
Thank you, Terrence.
I move it to IceBox