gw-vis/kagra-vis #39: Additional filters for the LVDT combiner.

MiyoKouseki, 2020-08-20:
[From here](https://paper.dropbox.com/doc/Plan-for-the-long-brake-towards-O3c--A6BHN4gn~Ey5XkTLymY8BCZaAg-KqexeqMaCA7DITFGxA5ih)

## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

Ryutaro-T, 2020-08-25:
Go to #74. This issue should be deleted.

MiyoKouseki, 2020-10-01:
Close it because it is duplicating to #74.

closed 2020-10-01 by MiyoKouseki