gw-vis/kagra-vis #48: Decide on suitable adjustment methods which should be applied for all suspensions and write documentation

MiyoKouseki, 2020-08-20:
[from here](https://paper.dropbox.com/doc/Plan-for-the-long-brake-towards-O3c--A6BHN4gn~Ey5XkTLymY8BCZaAg-KqexeqMaCA7DITFGxA5ih)

## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

MiyoKouseki, 2020-09-30:
@Ryutaro-T @naibafomsare do you know about this task?

naibafomsare, 2020-09-30:
The title of this topic sounds very general. We should talk about this with more specific ideas. We should do this in the VIS meeting or in a call.

Ryutaro-T, 2020-09-30:
This task came from items for O3c.
    - Oplev diagonalization: see the wiki for Type-B oplev page and klog 10786 for Type-A oplev.
    - IP LVDT diagonalization: several klog entries.
    - IP geophone diagonalization: klog 9064 and others.
    - IM OSEMs diagonalization: no reference.
    - Mirror actuation coil balancing: klog entries 13739 and 13614.