gw-vis/kagra-vis #50: Installation of the geophones or better ACCs in Type-A

MiyoKouseki, 2020-08-20:
## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

Ryutaro-T, 2020-08-25:
This task includes following items;

- [ ] Preparation of the LVDT coils. (Sato)
- [ ] Preparation of the LVDT drivers and coil drivers. (AEL)
- [ ] Replacement of the position sensors.
- [ ] Cabling of the new ACCs.
- [ ] Tuning of the folded pendulums.
- [ ] Test of the new ACCs.

MiyoKouseki, 2020-10-14:
@Ryutaro-T @NaohisaSato これは今どこで止まっていますか？

NaohisaSato, 2020-10-16:
@MiyoKouseki  コイル巻線作業（外注, 10/13納品）は終わりました。

NaohisaSato, 2020-10-19:
今後の予定
- 10/26-30 巻き線済みコイルを天文台で超音波洗浄、接着剤塗布。
- 終了後、ベーキング（外注）

NaohisaSato, 2020-10-28:
@Ryutaro-T 10/28　ベーキング前の洗浄、ポッティング終了。テクノポートに発送。

[20201026-28コイル洗浄メモ.pdf](https://github.com/gw-vis/kagra-vis/files/5450407/20201026-28.pdf)