gw-vis/kagra-vis #52: Installation of the limit switches for FR stepper motors

MiyoKouseki, 2020-08-20:
[from here](https://www.dropbox.com/s/60oygtwmckmdyj6/Towards%20O4%20on%20VIS%20.xlsx?dl=0)

## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

Ryutaro-T, 2020-08-25:
This task includes following items;

- [x] Design of the limit switches. (Sato)
- [x] Preparation of the limit switches. (Sato)
- [ ] Installation of the limit switches.
- [ ] Cabling of the limit switches.

AyakaAShoda, 2020-09-08:
Have you tested the switch?

AyakaAShoda, 2020-09-08:
- [ ] Prepare in-vac cables.