gw-vis/kagra-vis #54: Adjustment of GAS filter height in BS/SR3

MiyoKouseki, 2020-08-20:
[from here](https://www.dropbox.com/s/60oygtwmckmdyj6/Towards%20O4%20on%20VIS%20.xlsx?dl=0)

## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

NaoatsuHirata, 2020-10-01:
About BS Topfilter keystone height
- Fabian found that we need to reduce the ballast mass on the BS Topfilter keystone, but we don't have enough ballast masses on the keystone.
- At least we need to replace Topfilter disc to 50g lighter one. We can do it when we assemble yaw stepper motor.
- We should reduce more for contingency.
- Now I'm designing lighter disc.
[image]

NaoatsuHirata, 2020-10-01:
I uploaded design notes as below.
https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=12090

NaoatsuHirata, 2020-10-02:
uploaded drawings as below.
https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=12093

NaoatsuHirata, 2020-10-19:
I ordered parts. They will be delivered on 11/6 at Kamioka.