gw-vis/kagra-vis #56: Cleaning of the LVDT-ACT units in BS

MiyoKouseki, 2020-08-20:
[from here](https://www.dropbox.com/s/60oygtwmckmdyj6/Towards%20O4%20on%20VIS%20.xlsx?dl=0)

## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

Ryutaro-T, 2020-08-25:
Procedure of this task is as follows;

1. Dismount of the LVDT-ACT units from the pre-isolator.
2. Disassembly of the units.
3. Cleaning of the disassembled parts.
4. Assembly of the units again.
5. Mount of the units onto the pre-isolator.

MiyoKouseki, 2020-10-14:
@Ryutaro-T @naibafomsare Why do you clean the LVDT-ACT units in BS?