gw-vis/kagra-vis #57: Tuning of the IP in BS

MiyoKouseki, 2020-08-20:
[from here](https://www.dropbox.com/s/60oygtwmckmdyj6/Towards%20O4%20on%20VIS%20.xlsx?dl=0)

## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

terrencetec, 2020-08-24:
Hi, I didn't open this issue but I know the answer to your question. The IP of the BS was at around 65 mHz and was unstable. Mark and I went ahead to remove some arc weights so the frequency is now at somewhere above 0.1 Hz which is very bad for microseism. I imagine if we have good active damping system this wouldn't be a matter. However, having low res. frequency gives better margin of error, and, needless to say, better high-frequency noise attenuation.

MiyoKouseki, 2020-08-25:
Thank you, Terrence.
@Ryutaro-T san, this issue should be done in the current project by comparing human resources and time? Icebox or ToDo?

Ryutaro-T, 2020-08-25:
We try to tune the IP again when we open the top chamber. This is ToDo issue.

MiyoKouseki, 2020-08-25:
Thank you @Ryutaro-T san.

I have three questions;
- Did you decide when and who does this issue?
- There are some procedure documents?
- How do we define the review?

sorry, many questions...

Ryutaro-T, 2020-08-25:
This task is scheduled in the BS remedying work, and is applied by the Type-B/Bp team.
The procedure is very simple; adjust the small ballast weights measuring the transfer function of IP.
Review (acceptance) will be done by the transfer function.

MiyoKouseki, 2020-09-11:
Fabian checked the resonant frequency of Type-B suspensions. [klog#15013](http://klog.icrr.u-tokyo.ac.jp/osl/?r=15013)

He concludes that ...