gw-vis/kagra-vis #58: Fixing of the In-vac D-Sub connectors around the IP in ETMY

MiyoKouseki, 2020-08-20:
[from here](https://www.dropbox.com/s/60oygtwmckmdyj6/Towards%20O4%20on%20VIS%20.xlsx?dl=0)

## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

MiyoKouseki, 2020-09-30:
@NaohisaSato @Ryutaro-T これってどんなことする作業ですか？

Ryutaro-T, 2020-09-30:
The Dsub connectors around the IP are not properly fixed. They should be fixed by PEEK ties.

NaohisaSato, 2020-09-30:
In addition, attach the shell to the Dsub male connector of the PTFE cable. It avoids wire breakage. This is only for EY.
[Renovate EY In-vacuum cabling.pdf](https://github.com/gw-vis/kagra-vis/files/5303875/Renovate.EY.In-vacuum.cabling.pdf)