gw-vis/kagra-vis #59: Replacement of the damaged PTFE cable in ETMY

MiyoKouseki, 2020-08-20:
[from here](https://www.dropbox.com/s/60oygtwmckmdyj6/Towards%20O4%20on%20VIS%20.xlsx?dl=0)

## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

Ryutaro-T, 2020-08-25:
This task include following items;

- [x] Preparation of the new cable. (Sato)
- [ ] Replacement of the cable in the reinstallation work.

NaohisaSato, 2020-09-30:
About the damaged cable
- cable name:   CRYO7
- length:  6m
- where:  between BF and F2