gw-vis/kagra-vis #60: Replacement of the in-vac cable for the traversor in PR3

MiyoKouseki, 2020-08-20:
[from here](https://www.dropbox.com/s/60oygtwmckmdyj6/Towards%20O4%20on%20VIS%20.xlsx?dl=0)

## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

Ryutaro-T, 2020-08-25:
This task includes following items;

- [x] Preparation of the new cable. (Sato)
- [ ] Replacement of the cable.

MiyoKouseki, 2020-10-14:
@Ryutaro-T @NaohisaSato @NaoatsuHirata Are you ready to start this work after we open the chamber of PR3?

NaoatsuHirata, 2020-10-23:
At first, we need to investigate which cable was broken.
[image]

- Traverser has 4 cables for 4motors
- wiring as below
1: Blue, 2: Orange, 3: Black, 4: NC 5: NC, 6: Red, 7: Green, 8: NC, 9: NC
(confirm real wiring before work)