gw-vis/kagra-vis #61: Gluing of the suspension wire in PR3

MiyoKouseki, 2020-08-20:
[from here](https://www.dropbox.com/s/60oygtwmckmdyj6/Towards%20O4%20on%20VIS%20.xlsx?dl=0)

## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

MiyoKouseki, 2020-09-30:
@Ryutaro-T この進捗はどうですか？

AyakaAShoda, 2020-09-30:
Procedure: https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=11878
Takahashi-san and Hirata-san did some check at TAMA prototype payload with aluminum dummy TM.

MiyoKouseki, 2020-10-14:
@Ryutaro-T @AyakaAShoda Did we finish the preparation for this work? Is it ready to start after we open the chamber?

MiyoKouseki, 2020-10-29:
@naibafomsare As I told you, we should think about the distance between the pendulum and EQ stop after closing the chamber of TypeBp. We can get closer EQ stop to the pendulum because the Type-Bp suspension does not have IP which move the suspension point widely during the venting.