gw-vis/kagra-vis #62: Installation of additional beams for the outer frame

MiyoKouseki, 2020-08-20:
[from here](https://www.dropbox.com/s/60oygtwmckmdyj6/Towards%20O4%20on%20VIS%20.xlsx?dl=0)

## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

Ryutaro-T, 2020-08-25:
This issue is the same as #73, and should be deleted.

MiyoKouseki, 2020-10-01:
Close

closed 2020-10-01 by MiyoKouseki