gw-vis/kagra-vis #64: Modofication of the IMMT2 suspension

MiyoKouseki, 2020-08-20:
## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

Ryutaro-T, 2020-08-25:
This task includes following items;

- [x] Investigation of the present status. (AOS)
- [x] Design of the modified parts. (Sato)
- [ ] Replacement of the parts in the IMMT2 suspension.

NaohisaSato, 2020-10-20:
About parts procurement

- Drawings of suspension frame modification
https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=11918

- Sep 23 Ordered
- Delivery by the end of October

-

NaohisaSato, 2020-10-28:
Oct.28   Frame modification parts was delivered. 
We need to buy some screws for them.