gw-vis/kagra-vis #68: Decide on suitable adjustment methods which should be applied for all suspensions

MiyoKouseki, 2020-08-20:
[from here](https://www.dropbox.com/s/hoadb73fdyah02c/Post-O3%20task.xlsx?dl=0)

MiyoKouseki, 2020-10-01:
@Ryutaro-T @naibafomsare @AyakaAShoda Do you know what is this?

naibafomsare, 2020-10-01:
These are topics we need to talk more about, and include more people. As Takahashi-san pointed out in an email, these include (copied from his message ):

This task came from items for O3c.
- Oplev diagonalization: see the wiki for Type-B oplev page and klog 10786 for Type-A oplev.
- IP LVDT diagonalization: several klog entries.
- IP geophone diagonalization: klog 9064 and others.
- IM OSEMs diagonalization: no reference.
- Mirror actuation coil balancing: klog entries 13739 and 13614.

closed 2020-10-13 by MiyoKouseki