gw-vis/kagra-vis #73: New beams in the BS chamber (Mirapro)

MiyoKouseki, 2020-08-20:
[from here](https://www.dropbox.com/s/hoadb73fdyah02c/Post-O3%20task.xlsx?dl=0)

Ryutaro-T, 2020-08-25:
This task includes following items;

- [ ] Investigation of the present status at the site. (Mirapro)
- [ ] Design of the new beam. (Mirapro)
- [ ] Installation of the new beams.

MiyoKouseki, 2020-09-30:
@naibafomsare @NaoatsuHirata @Ryutaro-T 
How is the progress?

naibafomsare, 2020-09-30:
I have two comments:

- The beams are for the outer frame, not the chamber. Although many will understand it may be misleading for others.

- I don't think there is any progress on this. It seems Mirapro will come sometime to do measurements, but I don't know the details. It seems Uchiyama-san and Takahashi-san are managing this with Mirapro.

-