gw-vis/kagra-vis #79: New oplev covers

MiyoKouseki, 2020-08-20:
[from here](https://www.dropbox.com/s/hoadb73fdyah02c/Post-O3%20task.xlsx?dl=0)

Ryutaro-T, 2020-08-25:
This task includes following items;

- [x] Investigation of the present status and dimensions. (Fabian)
- [ ] Design of the cover. (AOS)
- [ ] Design of the pillar. (AOS)
- [ ] Setting of the pillars.
- [ ] Replacement of the covers.

MiyoKouseki, 2020-09-30:
@naibafomsare Do you know the progress?

naibafomsare, 2020-10-01:
This task has been assigned to AOS. Mirapro wll do it. Several week ago I asked Akutsu-san about recent progress and he told me there wasn't any. 

There is a document with guidelines at [JGW-E2011540](https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=11540).

MiyoKouseki, 2020-10-01:
Thank you. I understand.
@Ryutaro-T @AkutsuTomo Have we allocate time and human resources for that?