gw-vis/kagra-vis #86: Change the way to bind cables from Cryo Payload

MiyoKouseki, 2020-10-02:
EYにだけケーブルタイを替える。[資料](http://gwwiki.icrr.u-tokyo.ac.jp/JGWwiki/KAGRA/Subgroups/VIS/VISMinutes20201002?action=AttachFile&do=view&target=Cable+bandling+BF-CRYO.pdf)

MiyoKouseki, 2020-10-02:
[インターフェイスミーティング](http://gwwiki.icrr.u-tokyo.ac.jp/JGWwiki/KAGRA/Subgroups/VIS/VISMinutes20201002)

参加者：佐藤、高橋、牛場、阿久津、麻生、正田、木村、平田、三代

新方式にしたときのメリット・デメリット
 - Yaw方向のばね定数変わりそう。（どれぐらいゆったりできるか。）
 - インストールの効率があがる。

ケーブルのゆったり具合は改善する？
 - ロッドの中心ではなく外側で止めるのでその分余裕がある。

もし昔の固定方法に戻す場合、どれぐらい日数がかかるか？
 - 1週間かかる（ロッドから抜かないといけないので）

分割型にするほうがいいのでは？
 - これは時間がかかる

結論
 - 分割式にできるか考えるということに。

MiyoKouseki, 2020-10-14:
@NaohisaSato @Ryutaro-T これっていつまでに分割式にできそうか判断をするんでしたっけ？

MiyoKouseki, 2020-10-16:
[ミーティング](http://gwwiki.icrr.u-tokyo.ac.jp/JGWwiki/KAGRA/Subgroups/VIS/VISMinutes20201016)

YAW 方向の固定をすれば、OKということになった。

MiyoKouseki, 2020-10-18:
@NaohisaSato @Ryutaro-T 新しいケーブルクランプはいつまでに作って、いつまでにインストールする予定ですか？

NaohisaSato, 2020-10-19:
@MiyoKouseki @Ryutaro-T インストールはクライオパートがBFに接続される頃です（4月中旬）
下記スケジュールの＃88 Cabling がそれにあたると思います。
https://gwdoc.icrr.u-tokyo.ac.jp/DocDB/0114/L2011475/017/Schedule%20ETMY%20200807.pdf

NaohisaSato, 2020-10-20:
10月16日ミーティングの指摘（ケーブルクランプが回転しないようにしっかり固定する）を反映し、DRAFT図面をアップロード。
https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=12141

MiyoKouseki, 2020-11-04:
とりあえず、加工の発注が終わればこのTicketをCloseしたいと思います。
インストール時のTicketはまた近くなったら作ります。

NaohisaSato, 2020-11-05:
10月27日　加工発注（図面　https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument）
11月19日納入期限

MiyoKouseki, 2021-01-15:
納品されてインストールまち

closed 2021-01-15 by MiyoKouseki