gw-vis/kagra-vis #89: Check the location of the stepper motors on IP stage

MiyoKouseki, 2020-10-19:


MiyoKouseki, 2020-10-19:
## TypeA

## TypeB
### BS

### SR3 (Same as SR2,SRM)
https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=7077

[image]

MiyoKouseki, 2020-10-22:
@naibafomsare ( @terrencetec ) I tried to generalize your equation according to Fuji-kun's memo. Could you check the equation written in p.5 on [JGW-E2012144-v1](https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=12144)?

terrencetec, 2020-10-23:
The equation on page 5 looks about right to me. But there's a caveat. Let me just put a remark here even if it's not related to stepping motors.

We never used the dimensions from the CAD, except for dimensions that we cannot measurement by tape measure or caliper. This was suggested by Lucia a while back. They measure the LVDT positions and their angles and use onsite measurements to derive the actual sensing matrices that are used in Type-A suspensions. I don't now how they did they but later there was an occasion that proves the accuracy of this method: Nakano-san derived optical lever L2Y coupling using some IFO measurements, and we derived the exact, same value using the IP's displacement. This suggests Lucia's method is superior to those we used, which are CAD values. However, in Type-B suspensions, Enzo tried to replicate this method but seemed to have failed, according to recent measurements, which reveals huge coupling between IP DoFs. So, I had to do some ad-hoc diagonalization, whose result will not agree with your equation on page 5. 

As for the actuation matrix, and sensing matrix for geophones, we just intercalibrate or diagonalize them to the LVDT sensing matrix. Nevertheless, I am not sure how they fix the scale of the actuation matrix, it very possibly arbitrary.

naibafomsare, 2020-10-23:
- The reason the initial estimate of the matrices  was wrong was because Mark didn't read the correct value from the 3D-CAD. This was clarified later on.

- Measuring the position of the LVDTs/FR/Geophones in situ was fine, but it's worth mentioning the method proposed by Lucia was not particularly accurate. We just counted holes in the vacuum chamber and ticks we drew ourselves on pink tape put roughly in place by visual inspection. I checked the 3D-CAD later and the in situ measurements were roughly alright.

- The  matrix calculated by Miyo-kun is the initial estimate, not the final one for LVDTs and geophones. In those cases there has to be an additional process of diagonalization to take into account the asymmetries of the system. The final one and the initial one are not going to be the same.

In general, using the 3D-CAD is fine for fishing rods. For LVDTs and geophones is also fine provided it's just for an initial estimate.

MiyoKouseki, 2020-10-23:
Thank you.

Equation I wrote is the transformation matrix from local sensors (or actuators) to local axis defined by the location of the vacuum chamber in each suspensions. 

Of course, we have to rotate the matrix so that the local axis is aligned.

terrencetec, 2020-10-23:
> Thank you.
> 
> Equation I wrote is the transformation matrix from local sensors (or actuators) to local axis defined by the location of the vacuum chamber in each suspensions.
> 
> Of course, we have to rotate the matrix so that the local axis is aligned.

Just another remark, in general, the final transformation is not a rotation, it can be anything, as long as we can fix the scale of the outputs. For example, each LVDTs might measure a slightly different length scale and that cannot be fixed by a rotation.

MiyoKouseki, 2020-10-23:
OK. Of curse.

MiyoKouseki, 2020-10-26:
@naibafomsare

Fuji-kun's memo is here 
http://klog.icrr.u-tokyo.ac.jp/osl/index.php?r=7468

NaohisaSato, 2020-10-27:
@MiyoKouseki 方向Lの定義を教えてください。EYはEND方向と一致しているようですが、他の場所ではどうなりますか？ 
p.4のLとp.5のLは同じなのですか？
α1～3の起点は？

MiyoKouseki, 2020-10-27:
@NaohisaSato Slack

MiyoKouseki, 2020-11-04:
とりあえず、平田さんが修正した図面をみて、角度を確認しようとおもう。
https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=7077

MiyoKouseki, 2021-01-15:
Done. 
version6
https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=12144

closed 2021-01-15 by MiyoKouseki